AndyEveritt/ASMBL #37: MacOS Support

smoki3, 2020-09-02:
It seems that it is only running on Windows right?

AndyEveritt, 2020-09-03:
Unfortunately I do not have a mac to test on. There is no reason I can think that it shouldn't work on mac, you may just need to enable it in `ASMBL.manifest` potentially.

The standalone version would be the first thing to check if it works. The code doesn't use any non standard python libraries (for Fusion API reasons).

Would be happy to help you get it working on mac. Let me know how you get on

smoki3, 2020-09-03:
Hi Andy, 

I am able to load the plugin in fusion. but the output is not working.

This is what I get:
[image]

smoki3, 2020-09-03:
[image]

and also sometimes this

smoki3, 2020-09-03:
If I try to run the main.py I also run into this:

`kevin@Kevins-MBP-2018 ASMBL % python3 main.py
parsing files...
Opening files...
Converting additive gcode to relative positioning...
Spliting additive gcode layers...
Spliting subtractive gcode layers...
Ordering subtractive gcode layers...
Merging gcode layers...
Creating gcode script...
saving output...
Traceback (most recent call last):
  File "main.py", line 28, in <module>
    asmbl_parser.create_output_file(asmbl_parser.merged_gcode_script)
  File "/Users/kevin/Documents/3D Drucker/ASMBL/src/ASMBL_parser.py", line 343, in create_output_file
    os.startfile(file_path)
AttributeError: module 'os' has no attribute 'startfile'`

The result is an gcode file but it is not aligned

AndyEveritt, 2020-09-03:
Interesting. Would you mind exporting the f3d file from fusion. I haven't seen that issue before but if I can find the cause I can fix that next week. My guess is its an issue with the CAM setup

AndyEveritt, 2020-09-03:
Please can you also tick the `view intermediate files` in Fusion and send the `tmpAdditive.gcode` and `tmpSubtractive.gcode` in the folder it opens

AndyEveritt, 2020-09-03:
The alignment is done in `config.json` fyi

AndyEveritt, 2020-09-03:
Also very odd that the standalone works. They are fundamentally using the same code. Fusion just adds a GUI

AndyEveritt, 2020-09-03:
Also the os.startfile appears to not be a thing on mac. Will see if there is an alternative next week

smoki3, 2020-09-03:
[Gcodes.zip](https://github.com/AndyEveritt/ASMBL/files/5171553/Gcodes.zip)

these are the gcodes.

smoki3, 2020-09-03:
got it aligned with editing the config.json. So manual executing the main.py gives me now a working gcodes. Just still get this os.startfile issue. 

This is my test project
[ASMBL Test v8.f3d.zip](https://github.com/AndyEveritt/ASMBL/files/5171570/ASMBL.Test.v8.f3d.zip)

smoki3, 2020-09-03:
got this fixed by a simple condition:

`        try:
                if sys.platform == "win32":
                    os.startfile(file_path)
                else:
                    opener ="open" if sys.platform == "darwin" else "xdg-open"
                    subprocess.call([opener, file_path])`

AndyEveritt, 2020-09-04:
Thanks for doing the openwith fix. Will implement that when I'm back in office. It bizzare that the standalone works but fusion doesn't. Are you using the same 2 input gcode files?

smoki3, 2020-09-04:
Yes I use the same files.

I assume that the plugin is looking for some parameters in fusion? And may they are different between win and osx. 

I did some further testing and I was able to get some results without error message. But the merged gcode is broken. The additive part is just half finished

AndyEveritt, 2020-09-04:
That's very strange. I wonder if the post processors work differently on mac 🤔

AndyEveritt, 2020-09-04:
The other thought is to preview the 2 tmp gcode files in simplify. That will tell us if it is an issue with merging or the input files

AndyEveritt, 2020-09-04:
Just you try importing the spinner f3d from the examples folder and let me know if that setup works

smoki3, 2020-09-04:
So the 2 gcodes files (cam and additive) are looking good in simplifiy. Just the merged output is broken. Nevertheless as soon I do some more complex milling process it did fail with the issus from the screen shot

smoki3, 2020-09-04:
[image]

I get this when opening the spinner demo.

AndyEveritt, 2020-09-04:
That should be fine. The errors are just saying the operations are suppressed (leave them that way, I was trying some funky stuff) the warning is because the Retract height is too low for traditional milling... But we aren't doing traditional milling 😜

smoki3, 2020-09-04:
Okay then I get this again:

[image]

but if I merge the two gcodes manually with the main.py it looks like this :):
[image]

smoki3, 2020-09-04:
If I check the "Gernerate Toolpath" Option I get this:

[image]

AndyEveritt, 2020-09-04:
That's so frickin bizzare. I will try to look into it more next week. For now I would recommend using the standalone version since it seems to work.

AndyEveritt, 2020-09-04:
Also the generate toolpath button is buggy as hell due to bugs with the Fusion API, until they fix that it won't really work that well so probably best to avoid

smoki3, 2020-09-04:
Okay fine 👍 

If you need something more just tell me :)

AndyEveritt, 2020-09-04:
Sorry for the issues though . Hopefully we can resolve them fairly quickly 😊

smoki3, 2020-09-04:
No Problem :) I am here to fix it 👍

smoki3, 2020-09-29:
Have you made some progress on this?

AndyEveritt, 2020-09-29:
I was unable to recreate the issue my end. I did notice an error in your setup but don't think that it would cause the issue you were having,

In the `Linking` tab, you had a `Helix` ramp type
[image]

This should have been set to `Plunge` `No contact`
[image]

Its on my to do to warn users if they have not changed this themselves.

AndyEveritt, 2020-09-29:
My best guess is this logic in `src\fusion_api\Handlers.py` doesn't work on Mac and the individual gcode files are not created fully before the parser tries to combine them, do you think you would be able to have a play to see if you can figure out a solution as I still don't have a mac available to test.

```python
        try:
            start = time.time()
            postToolpaths(ui, cam, viewIntermediateFiles)
            while not (os.path.exists(tmpAdditive) and os.path.exists(tmpSubtractive)):
                if time.time() > start + 10:
                    ui.messageBox('Posting timed out')
                    return
                pass    # wait until files exist
            checkpoint = time.time()
            while True:
                if time.time() > checkpoint + 10:
                    ui.messageBox('Permission Error timed out')
                    return
                try:
                    open(tmpAdditive)
                    open(tmpSubtractive)
                    break
                except PermissionError:
                    continue
        except:
            ui.messageBox('Failed posting toolpaths:\n{}'.format(traceback.format_exc()))
            return
```

smoki3, 2020-09-29:
I will have a try this week.

Meanwhile could you merge this to fix the first issue on mac:

```
 try:
                if sys.platform == "win32":
                    os.startfile(file_path)
                else:
                    opener ="open" if sys.platform == "darwin" else "xdg-open"
                    subprocess.call([opener, file_path])
```

AndyEveritt, 2020-09-29:
That change should be in master already. Will create a new release later today

AndyEveritt, 2020-09-29:
https://github.com/AndyEveritt/ASMBL/releases/tag/v2.3.2

smoki3, 2020-09-29:
So I tested the new release in manual mode but this doesn't work on mac

```
Traceback (most recent call last):
  File "main.py", line 28, in <module>
    asmbl_parser.create_output_file(asmbl_parser.merged_gcode_script)
  File "/Users/kevin/Documents/3D Drucker/ASMBL/src/ASMBL_parser.py", line 333, in create_output_file
    os.startfile(file_path)
AttributeError: module 'os' has no attribute 'startfile'
```

smoki3, 2020-09-29:
Ah you forgot to add this to ASMBL_parser.py

smoki3, 2020-09-29:
So you was right. It a timing issue between the gcode generation and the merging process.

I added a `time.sleep(20)` before the merging process then it is working fine.

smoki3, 2020-09-29:
https://github.com/AndyEveritt/ASMBL/pull/41

AndyEveritt, 2020-09-29:
Cheers, if you are happy that it is working on mac now then I will merge with master and make a new release again?

smoki3, 2020-09-29:
Tested it now with a different models! Working fine so far. So I think you can merge it

AndyEveritt, 2020-09-30:
Just done a refactor, could you check it still works then I will merge to master

PR #42 (2020-10-07): Mac support
closes #37
closed 2020-10-07 by AndyEveritt