AndyEveritt/ASMBL #51: SuperSlicer not impressed by layerchange gcode

well-engineered, 2021-07-21:
As suggested, I did put 

`; LAYER_CHANGE`

to the layerchange gcode of SuperSlicer Version: 2.3.55
Build: SuperSlicer_
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 10.16 (Build 20F71)

SS does not put this string into its generated gcode.

If I change the layerchange gcode to something like:
`; "layerchange gcode" start:
; LAYER_CHANGE
;"layerchange gcode" end!`

the output results in:
`; "layerchange gcode" start:
;"layerchange gcode" end!`

I can imagine, that this is an SS bug/feature.

AndyEveritt, 2021-07-22:
I was not able to replicate this issue. It works fine for me on Windows. Maybe its a Mac issue?

I will change the layer change keyword to avoid compatability issues

AndyEveritt, 2021-07-22:
New flag needed to be added to slicer layer change script is `; ASMBL_LAYER_CHANGE_FLAG`

well-engineered, 2021-07-22:
New flag seems to work. I did leave the old flag in just to verify and this was the output:
[image]
[image]